markjfine/nrsc5-dui #3: Unable to load audio device

CaseyK9, 2021-04-01:
I get the error 21:16:15 Unable to open audio device.
when using it

Also isn't able to connect to my RTL, whilst the regular nrsc5-gui doesn't have any issues connecting.

Using Kubuntu 20.04

andrewfer000, 2021-04-01:
1. Make sure that no other SDR programs are running
2. Did you install all the python modules for python3 (`sudo pip3 install ...`)? Are you running `python3 nrsc5-dui` to start the program? 
3. Make sure that your default audio device is set.
4. Does nrsc5 alone work? 

I am using Ubuntu 20.04 with the KDE desktop which is the same as Kubuntu for the most part and I am having no issues. Somtimes, nrsc5-dui will crash on rare occasions and you need to run `killall nrsc5` in a terminal.

markjfine, 2021-04-01:
The regular nrsc5-gui uses the libraries to communicate with the RTL which causes audio problems. This one just runs nrsc5 as a separate process, lets it do it's thing, and periodically reads the status messages that it produces. So it needs to know where nrsc5 is to run it.

I hard coded the path to nrsc5 as /usr/local/bin/nrsc5. You might want to type:
`which nrsc5`
in the terminal to see where yours is installed and modify line 68 of nrsc5-dui.py accordingly. Or modify it to just 'nrsc5' and let your PATH environment variable do the work.

The reason I did this is because Python does some real strange things with paths when you try to run this using either Apple Script or via a Linux .desktop thing and a shell script (how I have it configured on a Mac and in Fedora). It will give me the same error you're getting with out the correct path.

I know that I need to automate this process better so it can find it in all of these situations. I'll make a note in my TODOs.

markjfine, 2021-04-01:
Should also add, you might want to check to see if you can run nrsc5 from the command line. If you can do that without getting an error, the app shouldn't produce one either. If you get the same error, make sure `sox` is installed, since I think the standalone nrsc5 process uses it to pipe the audio.

CaseyK9, 2021-04-01:
I'm going to call this fixed due to the fact that I was having issues with VMware properly adding my sound drivers. I'm going to make a new issue with feature suggestions and enhancements which I would really like to be added.

Thanks a lot for continuing to maintain this project.

closed 2021-04-01 by CaseyK9