markjfine/nrsc5-dui #4: Deprecation Errors

CaseyK9, 2021-04-01:
[image]

Something I'd like to see fixed or suppressed is the constant console spam from the Gtk libraries.

I usually leave the console open in the background in order to see and determine if there's any errors, but when this is constantly spamming me it would be nice to fix it.

markjfine, 2021-04-01:
I can appreciate that, and I'll make a note of it in the TODOs. I've tried to eliminate most where I've seen them, but Gnome's (Gtk/Gdk) recommendation is to just do all graphics from the main thread. So I need to look at that as well as the stock missing icon.

markjfine, 2021-04-01:
Actually they annoyed the crap out of me too, so I killed most of them. Only one left is the GtkSettings warning regarding the Cursor Theme, and that has to do with the GtkBuilder reading the glade file. When I do the code cleanup, I plan on dragging it into Glade to clean that up as well.

Am posting an updated new nrsc5-dui.py now.

andrewfer000, 2021-04-20:
Last time I checked this was fixed and this should be closed.

closed 2021-04-20 by markjfine