markjfine/nrsc5-dui #5: Fix Resolution On Album Art

CaseyK9, 2021-04-01:
[image]
Album art is constantly blurry, not too sure if you guys are already aware.

markjfine, 2021-04-01:
Cheers. Think if you make the window smaller it will resize it so it's not as blurry.
I'm going to be redesigning that tab completely, which will make the image even smaller. Might even poke a miniature version next to the station call up top.

markjfine, 2021-04-23:
Although the logos and related covers coming directly from the station are still 200x200, the album art coming from the MusicBrainz feature (replaces Discogs) is 500x500, so it should alleviate any resolution issues there.

markjfine, 2021-05-07:
Can this be closed?

andrewfer000, 2021-05-07:
At this point it should be closed. The original person who opened the issue is not responding and with the MB and Logo download features I think this should be considered fixed.