markjfine/nrsc5-dui #7: (Question) Does this support rtl_tcp

grumpzalt, 2021-04-22:
I run a personal rtl_tcp server to get my antenna in a better reception spot in my house, while still using RTL-SDR anywhere around the house, and i would like to know if this supports it.

markjfine, 2021-04-22:
Not at this time. Was just checking to see if nrsc5 even supports rtl_tcp. If it does, then it's just a matter of spawning nrsc5 with to appropriate argument to do so.

grumpzalt, 2021-04-22:
I have been using nrsc5 fine with rtl_tcp, you just need to use -H and put the ip address of the server.

Sent from my Verizon, Samsung Galaxy smartphone

markjfine, 2021-04-22:
Cool. Interesting... Let me see if I can't add something to the settings that would kick it off with that set of arguments.

I'm just not sure of the specfics of how nrsc5 works via rtl_tcp. Since it's remote, it will be interesting to see where the images and weather overlays are dumped to (locally? remotely?). Hopefully it tcps the files locally as well as the audio.

If I can't get it to work at least locally with 127.0.0.1, I may be coding this 'in the blind'. Are you willing to test it when ready if I can gets something ginned up?

grumpzalt, 2021-04-22:
Yea, I can test.

andrewfer000, 2021-04-22:
> I'm just not sure of the specfics of how nrsc5 works via rtl_tcp. Since it's remote, it will be interesting to see where the images and weather overlays are dumped to (locally? remotely?). Hopefully it tcps the files locally as well as the audio.

This should just send all the data from the SDR on the server to nrsc-5 on the client where the client does all the processing. This should not have any implications on code as long as nrsc5's rtl_tcp implementation works as expected.

markjfine, 2021-04-22:
Cool. So was able to get rtl_tcp running in the background using `rtl_tcp -a 127.0.0.1`, then was able to run `nrsc5 -H 127.0.0.1 103.3 0` to get WPRB. So, this is quite promising.

It's kind of cold and windy here... good as any a day to get my hands dirty on this. Should have something semi-operational in a few hours, though the status messages I'm seeing from rtl_tcp are a little confusing.

markjfine, 2021-04-22:
Updated... seems to work, including image files. #hooah

markjfine, 2021-04-22:
@grumpzalt: Give me a high sign if this satisfies what you needed and I'll close this (or you can - either way).

grumpzalt, 2021-04-22:
Yep, it's working fine with the text data and audio with rtl_tcp, no images yet but I am sure that's something to do with either the station or my signal. Thank you for responding and getting it working as quickly as you did!

markjfine, 2021-04-22:
No problem. I can't believe I overlooked something as important as this when you brought it up, if I'm honest.

closed 2021-04-22 by markjfine