markjfine/nrsc5-dui #8: Analog (analogue) Radio Implmentation [Long Term Goal]

andrewfer000, 2021-04-22:
Starting this issue in case someone starts working or wants to start working on implementing the following features and want's a good place to document problems and progress. This is somewhat extended from the original goal but not by much.

- [ ] Broadcast FM Support with RDS
- [ ] Broadcast AM Support with C-QUAM
- [ ] SW Radio support (would just require a change of frequency range of AM)
- [ ] NOAA Weather Radio (Narrow Band FM, 162-164Mhz)
- [ ] Scanning functionality

I think building it off of something like rtl_fm might be a good starting point. Since Nrsc5-DUI is just a frontend, it's better to have a way to do the demodulation and audio processing outside of the python program.

markjfine, 2021-04-22:
I'd also like to add DRM support as a counterpoint to ibiquity HD Radio at some point. Mostly for shortwave.

andrewfer000, 2021-04-22:
That's what i was thinking. Build or find smaller projects to get things to work and then integrate them into nrsc5-dui. Honestly, I think if we can just find projects similar to nrsc-5 but for am/fm and use Sox to get the information and control the software then it would be easier.

markjfine, 2021-04-22:
One of the guys responding said something about AM use. Clearly must be in DC because that's the only HD Radio AM station available 820 WWFD). I should tell him/her it's also a stream on both WTOP FM HDR stations (103.5 and 107.7). WWFD doesn't have the oomph to get down to where I am in Virginia, even with a longwire, but WTOP comes in just fine.

markjfine, 2021-04-24:
Correction. Just read that 1230 WFAS in White Plains, NY just applied to go all-digital. Hopefully that's not squashed by 1210 by you and you might have an opportunity to bag an HDR AM station using the app someday.

markjfine, 2021-04-24:
Check that... it's a 1kw daytimer. Good luck with that.

andrewfer000, 2021-04-25:
When I get home I'll check that. 1210 AM is pretty popular here in the Philly-Camden area so I doubt they'll be going all-digital any time soon. I can usually tune into them in my car or if I'm by a window using my RTL-SDR Blog v3 with direct sampling and the right antenna. I was never able to use nrsc5 to lock on to the HD signal (if it even has one still, it's still listed on hdradio.com though) Also, I don't know if my SDR can even pick it up, on my waterfall (CubicSDR) I do not see the digital side-bands like on FM.

andrewfer000, 2021-04-25:
Also I have been looking into possible easy tools to get FM Radio working on this program. I'm thinking rtl_fm might be our best bet. I don't know if that provides everything necessary in the way nrsc5 does but I could be a way to get FM (for broadcasts) and NFM (for NOAA). I'm not sure if you can decode RDS from the signal if it comes from rtl_fm either. This one will take some time and I wish we could get someone like the sdr-j-fm dev on-board to help with this (I know he's active since he implemented a fix for me in a day) and I think it might be our best bet.

markjfine, 2021-04-25:
Didn't mean to imply 1210 was digital, just that it might interfere with picking up 1230 from White Plains. All-digital AM stations don't have sidebands that straddle a main AM carrier like on FM. It's just one big digital chunk about the carrier freq. A normal AM radio just thinks it's noise and blows right by it.