markjfine/nrsc5-dui #10: Station Logos and Album Art Features

andrewfer000, 2021-04-29:
This issue will continue what has been going on in #6 only in regards to the development and testing of the Station Logos and Album Art Features

markjfine, 2021-04-29:
Got something working. Just need it to trigger something that goes past the first loop to see how it fully works. The station I'm on is keeping the logo on for all ones I typically had a problem getting cover art for - making that hard.

What I did was set the type and status look for variables that are set to 'Album' and 'Official' when appropriate, and to '*' when not, all based on the value of i.

Think what I'll do as a compromise is change the checkbox I currently have for Strict in Settings to be something for Extended Searches instead. This will give people an opportunity to turn go past the first loop when enabled. Basically make it an option.

andrewfer000, 2021-04-29:
Great! I'll test it out as soon as the code is updated.

markjfine, 2021-04-29:
Ok... it's up there. Just so you know, I actually find the non-strict queries to be very slow. In fact, I'm getting the old Mac beachball waiting for some of them to finish. I don't recall them being that slow before so I'm now wondering if we're not getting throttled.

markjfine, 2021-04-29:
crap... think I found a bug in the recording-list loop. Seems like it's never getting past the first recording

markjfine, 2021-04-29:
n/m

andrewfer000, 2021-04-30:
So I'm testing your changes and they work okay. For some songs, the 8 filters were able to catch things that this one is not. I need to test it for a little longer to see if the lower query trade-off is worth it or not. I defiantly think adding the manual add album art would be a good idea (maybe do it like the station art, but have it so when the user right clicks on the Title a textbox will appear for the link to go into). I honestly think this is as good as it's going to get for awhile unless someone from MusicBrainz comes to help refine our queries (I believe it can be improved but we just don't know what makes MB "click"). Let's focus on refining _how_ album art and stations logos are saved and managed for now and let's re-vist this part later.  At this point we did everything we could for the time being (I mean seriously _I_ wrote code lol). Let's maybe make the next goal be making downloaded logos work for any station. It'll probably be less annoying than trying to work with MB xD.

andrewfer000, 2021-04-30:
Yeah I am running into some issues with the new album art system. What does "lastXHDR: 1" mean and why does it prevent the lookup of songs at times?

markjfine, 2021-04-30:
That's what's preventing the search to occur when the station goes to the logo. Unfortunately, it also appears to be sent when they don't transmit album art. So we have two choices:
1. Continue to prevent searching when the station sends an XHDR of 1.
2. Search for everything including text during logos and default to the logo whenever nothing's found (this was the way I originally had it for download mode).

Quite frankly, I don't see how we can get around not using the second option to do this correctly. The text really should be driving what is being seen in this mode. I would also feel a lot more comfortable if we chopped the number of searches down so there's a 76% solution instead of trying for a 100% solution. For example, if 1 and 4 will get you most of what you want to see, let's just do 1 and 4, or 1 and 8 (if that widens it up enough). The fewer non-Strict searches done will also cut down on the search time and number of hits, which will be beneficial in the long run.

markjfine, 2021-04-30:
I should also add that there are stations that don't provide album art but do provide track info. In these cases you don't get an XHDR message so it defaults to " ", and never "1", so it's doing the search on logo text anyway.

andrewfer000, 2021-04-30:
I'd personally prefer a toggle for this feature. Once I get most of the artwork downloaded, I can turn the XHDR thing back on but in the beginning I'd like to keep it off. I think that's the best solution of them all.

closed 2022-12-19 by markjfine