markjfine/nrsc5-dui #16: Possible to run without an audio device?

salamanders, 2022-01-23:
I'd like to capture to files, and the dusty machine nearest to where I can plug in the antenna doesn't have an audio device.  Is there a way to configure the app to only dump to audio files?

I did get it to compile and run in Ubuntu, and got as far as 

```
$ more nrsc5.log 
15:32:17 Unable to open audio device.
 ```

markjfine, 2022-01-24:
This application is intended to be a simple player that just calls the nrsc5 executable in a shell.

I know you can read an iq file into it and save the decoded wav output, but I thought nrsc5 still required an audio device to run.

You might be better off asking the nrsc5 developers (theori-io/nrsc5) if there's a command line method of ignoring the presence of an audio device. I'm not really certain if there is. If it's not available, perhaps that's a feature they can add.

salamanders, 2022-01-24:
I was able to run nrsc5 from the command line with the output going to a file, so I don't think the binary absolutely requires an audio device.  But my use case is very corner-case - I'm betting most of the ppl who run this app do so to play audio!