markjfine/nrsc5-dui #17: Installing nrsc5-dui on Windows 11

FairyballYT, 2022-01-29:
Hi, I understand that at this point of time, there isn't an official .exe version of nrsc5-dui for Windows computers, so yesterday I tried to compile the source code for nrsc5-dui in MSYS2 MinGWx64 and MSYS2 MinGWx86. I keep running into this problem where the setup.py file can't find a module called gtk+ and I know that I have gtk2 and gtk3 installed in MSYS2 MinGWx64, so I'm not sure how to fix this error message.
[image]

markjfine, 2022-01-29:
There's a lot to unpack here...

1. Until there's such a thing as a Python compiler, there will never be an 'exe' for scripts like nrsc5-dui (or nrsc5-gui). So, I have no idea how you were able to try "compile the source code". You were just running it under Python, as you should be.

2. This Python script has several dependencies which must be installed prior to running. Those were outlined in the README.md. That said, setup.py is a holdover from another version of nrsc5-gui and I'm not sure of it's value for this version. If you want to continue experimenting, you should concentrate on running nrsc5-dui.py.

3. Finally, and more importantly, I recommend you read the entire Windows 10 Setup Notes in the README (which should also apply to Win11) as well as the full Windows Support Thread [https://github.com/markjfine/nrsc5-dui/issues/9] before continuing. There are several show-stoppers that I ran into getting this to work in Windows, one of them being that a cross-compiled nrsc5.exe has issues working properly even on it's own.

FairyballYT, 2022-01-30:
Thanks for the reply, I also tried to launch nrsc5-dui.py in MSYS2 MinGWx64 but MSYS2 thinks that the file doesn't exist, strange. It seems like you're right, I was just trying to run the program under python, but there are a few modules that I either haven't installed yet, or are there, but MSYS2 isn't detecting. I did try to install every dependency listed in the README.md file, but even then, it won't run on Windows. Thanks for the help, I'll stick to using someone else's pre compiled version of nrsc5-gui for the time being. 
[image]
 
[image]

markjfine, 2022-01-30:
It couldn't find it because you typed it wrong.
Also, 'pre-compiled'? Which version are you saying is 'pre-compiled'?

FairyballYT, 2022-01-30:
> It couldn't find it because you typed it wrong. Also, 'pre-compiled'? Which version are you saying is 'pre-compiled'?

https://github.com/zefie/nrsc5-gui

andrewfer000, 2022-01-30:
> There is a way to package Python programs into .exe files. I've never used it before but nrsc5-gui does have it. If we were to get this running on Windows i've would have done it.