markjfine/nrsc5-dui #19: Feature Request/Dark Mode

ferrellsl, 2022-02-17:
Would it be possible to implement a dark mode?  The white window background can be particularly eye-searing in dim settings.  Maybe a black background with light gray text.  Thanks for all the work you've put into this app. I've enjoyed it immensely.  I set aside an entire mini-HTPC as my HD Radio infotainment center and placed it in my kitchen.  

[image]

markjfine, 2022-02-17:
The light or dark mode should follow the operating system settings, however, I think I've noticed this might not be the case in Linux. I know this to be true in macOS.

I'll look into it. Might be a python setting somewhere as well.

markjfine, 2022-02-17:
Lol. No worries. Enjoy.

closed 2022-02-21 by markjfine