markjfine/nrsc5-dui #20: Simple Fix for Windows users/WSL2

ferrellsl, 2022-02-21:
I found a very simple fix to get your app running under Windows/WSL2 without any issues.  WSL2 accesses the Windows file system from the /mnt folder.  I.e., to access files on the Windows C: drive, provide the following path:  /mnt/c and to access the D: drive, /mnt/d/ and so on.

By editing line 74 of nrsc5-dui.py and adding the path for the nrsc5.exe and libnrsc5.dll, the app runs fine.  The edited line reads:  self.nrsc5Path = arg1+"/mnt/d/nrsc5/nrsc5.exe"

My EXE and DLL are in my D: drive in a folder named nrsc5.  For users who have their nrsc5 files in another location, simply edit line 74 to reflect where your files are stored.

I have to invoke the app as root using sudo, like so:  sudo python3 ./nrsc5-dui.py
Not sure why, but without sudo, python will throw an error.

Here are some screenshots:

[image]

[image]

[image]

markjfine, 2022-02-21:
Can edit line 74, but you should be able to just add the path to nrsc5 to the command line.

ferrellsl, 2022-02-21:
After some more testing I have encountered one problem with this method.  When I click on the channel tabs for a particular frequency, the channel does not change even though the channels are identified correctly.

markjfine, 2022-02-21:
That's likely because the environment isn't Posix compliant. The app is supposed to pipe a channel number to the nrsc5 thread, but the pipe won't work correctly in a non-Posix environment. There's a replacement method for use under Windows (WinPty), but it would be a major rewrite.

ferrellsl, 2022-02-21:
I found a workaround for changing channels that isn't elegant but it works.  To change to another channel, I have to double-click its tab.  Then press the stop button.  Then press play again, and then I'm on the correct channel  Not a big deal for me.

markjfine, 2022-02-22:
Can just click stop, then click the tab, then click start. Basically the same thing.

markjfine, 2022-02-22:
Disregard. Forgot I disabled the tabs when stopped.

ferrellsl, 2022-02-22:
No worries.  Thanks for all the work you put into this.  I bet it's almost a fulltime job.

markjfine, 2022-02-22:
Not really now, but when I was developing it felt like it at times.

ferrellsl, 2022-02-22:
Just another little tweak.  WSL2g doesn't install complete themes when you enable graphic applications such as Gedit or Nautilus so NRSC5-DUI will complain about missing icons such as the Yaru 'help-about' icon and some default mouse pointers.  Apparently WSL2g defaults to the Yaru desktop theme which is fine but you will want to install the missing portions of the theme by typing:  sudo apt-get install yaru-theme-icon

After doing so you can now see the large star icon on the menu bar which will bring up the "About NRSC5-DUI" dialog box that was inaccessible until after updating the theme.

[image]

ferrellsl, 2022-02-22:
Hi Mark.  I have a question for you.  I noticed that I can double-click on a station/channel#  in my bookmarks window and move from station/channel seamlessly without having to hit the stop/play button.  For example, I can be listening to 95.5-1 and then double-click on 101.5-3 and it works perfectly, no need to hit stop, select the next station and then hit play.  I was wondering if the tabs on the menu bar could be modified to use the same tuning method that's used in the bookmarks window?  If so, this would get this WSL2g option up to 100% with the native Ubuntu or MacOS options.

I say 100% because my hacked version of your app even shows the correct MER/BER rates, so the tabs issue I mentioned is the only thing left that requires me to use a workaround.

I'm not a python programmer so I may be asking for the stars here but I thought I'd ask anyway.

markjfine, 2022-02-22:
Because of the way the command loop is written, I'd rather have it use WinPty to try and pipe the correct channel number (0-3) and quit (q) commands to nrsc5. Those are quicker and cleaner than stopping the nrsc5 loop and having it re-synch the station on a restart. Failing that, I'll use your suggestion as a backup in order to get it working in Windows.

A precursor to all of this (including the addition of a user-defined nrsc5.exe path to the Windows command line, as noted earlier) is getting PyGObject to install properly. Still having an issue with that under MSYS2, which it should also be able to run under. I'd hate to give up on MSYS2, but will move to WSL2 if necessary.

I'm also running Win11 under a Parallels virtual machine that apparently has USB comms/timing (and possibly processing) issues, so am battling that lovely 'feature' as well. Almost thinking of getting a semi-cheap Windows box just so I can play some Steam games (some games won't run due to an limited graphics capability on my early-2013 Mac and I absolutely refuse to buy a new Apple Silicon).

ferrellsl, 2022-02-22:
Hi Mark.  PyGObject is installed properly on WSL2g.  WSL2g is an Ubuntu 20.04 Linux VM that is hosted by Windows just like Linux under VMWare on Windows.     Unfortunately, WSL2g doesn't support audio or USB pass-through out of the box yet.  With some hackery you can get USB passthrough working by installing usbipd [https://docs.microsoft.com/en-us/windows/wsl/connect-usb](url) on Windows but I found it very buggy. [https://www.xda-developers.com/wsl-connect-usb-devices-windows-11/](url)

A better option that worked reliably for me was VirtualHere and it has a GUI.  [https://www.virtualhere.com/usb_client_software](url)  It worked perfectly but then there was the audio problem on WSL2g.  Because there is no virtualized audio hardware under WSL2g you have to install a pulseaudio server on the Windows side which works for most Linux apps, but unfortunately, NRSC5 doesn't work with pulseaudio.  It wants to use real hardware audio via LIBAO, so it will fail to run under WSL2g reporting that it's unable to open the audio device.  That's why I modified line 74 of nrsc5-dui.py.  This allows Windows and the nrsc5.exe to handle the USB and hardware audio while all the Linux dependencies and scripting are handled by WSL2g via Ubuntu 20.04.

I also have your app installed on a Linux VM under VMWare on Windows 11 where it works perfectly.  VMWare supports USB-passthrough and hardware audio so nrsc5-dui and the nrsc5 binary run great there.  I suspect that both would run fine under VirtualBox too, but I haven't tried it yet.  Since VirtualBox is freeware, I will test it today and get back to you.  I think most users will prefer to use WSL2g in the long-term though as it works seamlessly within Windows (minus the aforementioned problems which Microsoft is addressing) and doesn't require the user to install and fire up a VM host first as with VMWare or VirtualBox.

markjfine, 2022-02-22:
Good info to have.

ferrellsl, 2022-02-22:
I just finished testing nrsc5-dui under Ubuntu 20.04 on VirtualBox.  Everything works great.  There was an issue with video during installation of Ubuntu.  You have to make sure that 3D acceleration is enabled and you have to max out the video RAM or you won't get a login screen.

[image]

Once Ubuntu was installed properly it was a quick matter to install nrsc5 and all the python dependencies for nrsc5-dui.

One other minor issue that occurs under VirtualBox is that the audio will sound scratchy when moving the mouse while nrsc5-dui is playing.  This doesn't change even when plugged into a USB3 port with USB3 pass-through enable.  Other than the aforementioned, all works fine.

ferrellsl, 2022-02-22:
I was able to get rid of the scratchy audio by changing the VM's mouse from USB to PS2. So VirtualBox is a great solution for Windows users who don't mind fiddling with VrtualBox and virtual machie=nes.

[image]

ferrellsl, 2022-02-22:
For Windows/WSL2 users who would like a simple CMD file that loads nrsc5-dui from a Windows command prompt there are a couple steps that will get you there.  First, use pyinstaller to compile nrsc5-dui into an executable file.

Open your WSL2 command prompt and navigate to your nrsc5-dui folder.  On my system it's located at ~/nrsc5-dui
Once there, run pyinstaller by invoking:  pyinstaller nrsc5-dui.py.  This will leave an nrsc5-dui binary in the ~/nrsc5-dui/dist/nrsc5-dui folder.  Make sure to create the following folders in the ~/nrsc5-dui/dist/nrsc5-dui folder:  aas, map, cfg, and res. 

Next open a text editor in Windows and add the following line:
wsl bash -c "cd /home/steve/nrsc5-dui/dist/nrsc5-dui;echo "your-password-here" | sudo -S /home/steve/nrsc5-dui/dist/nrsc5-dui/nrsc5-dui;"

Make sure to edit the line and replace the "your-password-here" with your WSL2 root password leaving out the quotation marks. Replace the "steve" instances with your user ID.

Save the file on your PC as nrsc5-dui.cmd.  I saved mine in the top-level directory of my C: drive.
You should now be able to run the nrsc5-cmd file from powershell or a Windows command prompt and it will start the nrsc5-dui  app without any further work. i.e., C:\nrsc5.cmd

You can also double-click the command file to start nrsc5-dui.

If you experience random permissions/ownership settings being changed on your folders in WSL2, you should create a file under /etc and name it wsl.conf

Add the following lines to wsl.conf and save the file:
 [automount]
 options = "metadata"

Next, reboot your WSL2 virtual machine by opening a CMD or powershell prompt and typing:  wsl.exe --shutdown
Now you can re-start WSL2  and any permissions or ownership changes that you make thereafter should stick.  I had to do this because root kept taking ownership of the aas, maps, cfg, and res folders that I created under ~/nrsc5-dui/dist/nrsc5-dui