markjfine/nrsc5-dui #34: I got a couple questions.

BenHut1, 2023-06-13:
First question. Can you add an option to this program to load a WAV file as a source? I used HDSDR in upper-sideband mode and with output bandwidth at 192kHz which gives me 96kHz (just enough for one of the 2 copies of the HD Radio signal, as each FM station transmits 2 copies of the HD Radio signal), and saved some of this signal to a WAV file. I would like to be able to load a copy of this WAV file as input to your software, so it can output a WAV file of the decoded audio.

Second question. Are the upper and lower copies of the HD Radio signal (one above the FM carrier and one below the FM carrier) just 2 copies of the same signal for redundancy? Or do they carry separate data streams, thus requiring an even wider bandwidth, and the decoding of both of them, in order to get the actual audio signal?

andrewfer000, 2023-06-13:
Hi Ben, 

I feel like you'd get a better answer by asking this question on the nrsc5 GitHub since those are the folks who developed the decoder. NRSC5-DUI is just a graphical front end for the nrsc5 command line program linked in the ReadMe.

markjfine, 2023-06-13:
...What he said. 😀

markjfine, 2025-06-05:
FYI - I started looking into how to read sdr-recorded IQ files. I just need to design a file dialog to select the file and allow you to set the frequency it was recorded from.