markjfine/nrsc5-dui #35: [macOS] Album Art does not display in the UI.

ControlAltAndDelete, 2023-07-20:
In the title. For some reason even if musicbrainz or a stations album art is downloaded and is present in nrsc5-duis files, in macOS, the UI does not display it. This problem does not happen in Ubuntu Linux though. I don't know if its my packages but I did use pip to install the requirements.txt. nrsc5-dui 2.2.1 is the last version able to display album art in the UI. This also affects the traffic and weather map features as well.

markjfine, 2023-07-20:
I'll check on the musicbrainz issue.

I'm pretty sure display of traffic and weather maps are a different issue completely. Changes were made to the most recent version of nrsc5 that affected the detection of all traffic and weather map images, as well as any station-provided logos and album art. Those changes were recently accounted for in the code but hasn't been included in a release because it hasn't been fully tested. You can remedy the traffic/weather issue by pulling down the latest version of nrsc5-dui.py from the Code listing. I doubt this will fix musicbrainz, but it's worth a shot.

ControlAltAndDelete, 2023-07-20:
[https://drive.google.com/file/d/1a74E_FNjYRp-X5Ls587gQyLFtkUc9bMP/view?usp=sharing](url) the new build didn't work even the one where I used the code button to download everything. it is recieving radio station data but seems it can't display it in the nrsc5-dui app?

markjfine, 2023-07-20:
And you have the absolute most recent nrsc5?

ControlAltAndDelete, 2023-07-20:
Just updated it and everythings fixed now apart from musicbrainz. It's strange because downloading album art from musicbrainz worked on ubuntu before and pulled it everytime adding the album info along with it, don't know why it doesn't work now though.

markjfine, 2023-07-20:
Dunno. Just tried it and works here. Check the privileges on your aas folder.

ControlAltAndDelete, 2023-07-20:
Yeah their okay. Album art, weather map, traffic map, station logos show up now. Only problem now is musicbrainz... However, the bookmarking feature has a bug. If you bookmark a station, changing to the HD2/HD3/HD4 disables bookmarking so, you have to hit stop and then play to get the bookmark button back. Is that a bug?

markjfine, 2023-07-20:
Yup... Noted.

markjfine, 2023-07-20:
Also, I know what requirements.txt says, but these things are constantly updated. Might want to update each in pip. Here are the versions I am currently running. Homebrew updates most of them for me:
pygobject==3.44.1
musicbrainzngs==0.7.1
numpy==1.25.1
Pillow==10.0.0
pyOpenSSL==23.0.0
python-dateutil==2.8.2
urllib3==1.26.14

markjfine, 2023-07-21:
If you're running from a terminal, check to see if there are any python errors posted and let me know.

You might not see this on WTDY, but I've noticed the musicbrainz query can fail depending upon how the station lists a song title or artist. For example, certain words like 'featuring' can be listed as 'feat.' or 'ft.' or even just 'f'. If it doesn't match what musicbrainz expects then it will return no matches. It's a very common problem on hip-hop stations. You can try setting the 'Extended Queries' toggle in Settings to see if that helps. I don't recommend setting this for laggy internet (like satellite) though as it will slow everything down.

It's also possible that the station-provided album art is taking priority over the musicbrainz query. In that case, you can try turning off the 'Include Station Art' toggle in Settings.

ControlAltAndDelete, 2023-07-21:
musicbrainzngs is just somehow crapped out for me, made sure everythings up to date, not sure if its me, or their server because when I tried nrsc5-dui on ubuntu, musicbrainzngs was able to download album art, and the album, it's not working on my macbook. 

This is the output everytime: 
(nrsc5-dui.py:997): Gtk-WARNING **: 09:19:45.261: Could not load a pixbuf from /org/gtk/libgtk/theme/Adwaita/assets/check-symbolic.svg.
This may indicate that pixbuf loaders or the mime database could not be found.
['nrsc5', '--dump-aas-files', '/Users/pauldelfierro/Downloads/nrsc5-dui-master/aas', '96.5', '0']
MusicBrainz recording search error

and the station was playing an easy song that the musicbrainzngs should be able to pick up, it did on ubuntu but not mac :/
Artist: Taylor Swift
Song: Cruel Summer

ControlAltAndDelete, 2023-07-21:
Unrelated but, an interesting tidbit I found is that apparently WKXW-HD in Trenton, NJ somehow managed to pull off up to around ~83 KB ish audio. It's doing that while broadcasting the NAVTEQ stuff that is displayed on garmins and car GPSes. I think they might be abled to pull this off with no other subchannels but the engineers managed to pull it off. 

** Also, implementing HERE/NAVTEQ traffic might be very difficult or probably impossible because it's not an image, it's a lot of data that gets sent to a car and layers it onto its own map. it's very different from traffic map and it sends a lot of data over to the car so a way would be needed to maybe pull up an interactive openstreetmap and layer on the lines of traffic along with icons for incidents, construction, traffic regulation changes.

*** I wonder if more weather data can be pulled and if its from iHeart. The local forecast feature could be a bit less difficult than the whole NAVTEQ thing but you'd have to find the icons and it'd take a long time to put in tabs for Weather as it pulls in your location, local cities, and other cities. It also creates more time needed to implement a way (possibly via IP address) to triangulate the location to display weather for a single location.

The first ones just something interesting and the ** and *** are just some talk about the technology sent. There's also apparently "fuel prices" and "News Headlines" are sent via HD Radio but I doubt anyone looks at those and if they even send those anymore. Heck, it's not even in any of my cars, well TOYOTA never put them in just, NAVTEQ, Doppler, and iHM traffic map for units without navigation.

markjfine, 2023-07-21:
Not being able to find check-symbolic.svg indicates it's missing a toolbar icon. Shouldn't affect musicbrainz at all.

I'm away from my machine but I'll check on how to fix that as well as the MusicBrainz recording search note. Just a thought, but it may be something silly like a UTF-8 mismatch.

markjfine, 2023-07-21:
Regarding NavTeq/HERE: The latest nrsc5 currently handles it, but only through the API at the moment. Once Clayton's certain it works he plans to add a command line switch similar to the download-aas thing. I was able to implement it easily in a version of nrsc5-gui while it was being debugged, so this will be coming at some point.

I should add that the traffic maps are nice and clean, but the weather maps are much more narrowly localized than what iHeart provides. The background image currently used for iHeart weather doesn't lend itself very well so I may look for a different way to get a background map to put the HERE weather overlay on.

markjfine, 2023-07-21:
Regarding check-symbolic.svg: If I remember correctly, I think you need the homebrew package adwaita-icon-theme and to have gtk+ use it as the default theme in ~/.config/gtk-3.0/settings.ini:
```
gtk-icon-theme-name = Adwaita
gtk-fallback-icon-theme = hicolor
gtk-cursor-theme-name = Adwaita
gtk-cursor-theme-size = 0
gtk-toolbar-style = GTK_TOOLBAR_ICONS
gtk-toolbar-icon-size = GTK_ICON_SIZE_SMALL_TOOLBAR
gtk-button-images = 1
gtk-menu-images = 1
gtk-enable-event-sounds=1
gtk-enable-input-feedback-sounds=1
```
Otherwise, I think I may have just copied the .svg from Fedora into a place it would see it, e.g., /opt/homebrew/Cellar/adwaita-icon-theme/44.0/share/icons/Adwaita/symbolic/ui/check-symbolic.svg

Regarding the MusicBrains error: Looks like either the search timed out, or more likely that macOS is not letting python use the internet to perform it. Might want to check Settings/Network/Firewall and see how it's configured.

ControlAltAndDelete, 2023-07-22:
[image]
hm, musicbrainz is still failing to get the album art for some reason, firewall is off too so I don't know what it could be. (btw artist experience and the station logo does show up, it just didn't download yet when I took the screenshot)

markjfine, 2023-07-22:
Judging from the error, I should have mentioned that the first line in settings.ini should be "[Settings]". I just assumed you had that set. Also, since you are apparently running dark mode, you can also add the 2 lines I've included. All of this should be at the very beginning of settings.ini:
```
[Settings]
gtk-application-prefer-dark-theme = true
gtk-theme-name = Adwaita-dark
```

I've also just added some debugging to nrsc5-dui.py to detect what it's sending to search. Maybe that will give me a clue to what's happening. If you re-pull nrsc5-dui.py it should show the next time you run it.

ControlAltAndDelete, 2023-07-23:
MusicBrainz recording search error
iteration = 1 .
imgSaved = False .
strict = True .
artist = Doja Cat .
recording = Say So .
type = Album .
status = Official .

this happened on Q102-HD3 "Pride Radio" as its the only station that I can quickly pull in to get the artist and song info for the result

I confirmed its in their database too: https://musicbrainz.org/release/56f19716-e7c5-486a-88ec-8e2ceaff1c8a

markjfine, 2023-07-23:
Yeah, but it quit straight away. This is still looking like macOS blocking Python from using the internet.
Double check that under Settings->Network->Firewall->Options... that:
1. "Block all incoming connections" is NOT checked;
2. If there's something that looks like it should be listed for Python in the list, that it's set for "Allow incoming connections";
3. "Automatically allow built-in software..." is checked; and
4. "Automatically allow downloaded signed software..." is checked.

ControlAltAndDelete, 2023-07-23:
[image]
its still somehow happening and I've tried it with both FileVault disabled and enabled and its still happening I don't know if there's any terminal commands I can try

markjfine, 2023-07-23:
Yeah, this is weird. Have to admit I'm stumped.

ControlAltAndDelete, 2023-07-23:
ill update you later tonight when I reinstall ubuntu on an hp envy I have, though to my prior knowledge, it did work under ubuntu

ControlAltAndDelete, 2023-07-23:
yeah I am just not sure, it downloads every time in ubuntu with album and everything. is it something with Ventura or something?

ControlAltAndDelete, 2023-07-23:
[image]
I did notice this weird quirk where even if musicbrainz successfully downloads album art and the album name it still spits out some error even though nothing is affected on linux

markjfine, 2023-07-24:
When it does a query it is actually a list of possible things. If one fails it goes to the next one. Sometimes the image might be unavailable for some reason or it was mis-linked. Saw this on the web site as well, which is why I let it bounce to the next possible one.

ControlAltAndDelete, 2023-07-24:
[image]
how is it missing in /opt? is it an access issue because even it is not seen in the terminal even with sudo

markjfine, 2023-07-24:
Or, wherever Homebrew is installed. I was of the understanding that Homebrew needed to be under /opt on newer ARM machines.

ControlAltAndDelete, 2023-07-24:
[image]
it has openssl but not pyopenssl

markjfine, 2023-07-24:
I think I know what's going on here.

It appears you're not using Homebrew (or similar), but it's confusing because it looks like you have Python 3.11 installed along side of macOS's stock version of Python 3.9 (python3) in /usr/bin. Just based on the Python naming, I'm also guessing you're running an Intel and not an M1/2 ARM Mac, which at a minimum is why there'd be nothing in /opt.

Likewise, the openssl you're likely using is the stock version that comes with macOS, which is actually LibreSSL and has a few TLS compatibility problems. It's not really openSSL, but a fork of it that goes back to 2014. My macOS reports /usr/bin/openSSL as LibreSSL v3.3.6. Homebrew installed the latest version of the real openSSL this past May, v3.1.1, and is 100% TLSv1.3 compatible. I'm pretty sure that Ubuntu also uses a fully compatible version of openSSL, not LibreSSL, which is why it works there. I've always had problems with macOS's stock openSSL, which is why I installed the one from Homebrew in the first place.

All this said, I have no idea how to test this theory, because disabling openSSL will also disable my running/working version of Python and won't fully isolate the issue. You could go on Ubuntu and type "openSSL version" and see what it comes back with. I'll bet it says something like "OpenSSL 3.1.1 30 May 2023".

Regarding pyopenSSL... that goes wherever Python's pip3 installs it. You can see if it's actually installed by typing "pip3 list" and seeing if pyopenSSL and cryptography are installed, along with the other Python packages you have installed.

markjfine, 2023-07-24:
One more thing, if you do install the actual OpenSSL, it's probably not a good idea to overwrite the one macOS provides. It may work until macOS gets updated again (which is almost weekly now) and overwrites your good version with LibreSSL again.

ControlAltAndDelete, 2023-07-24:
do you have any guidance on how to redirect these calls to MusicBrainz through openssl instead of libressl? I'll try any command and it doesn't matter if I break my OS as i'll just reinstall and everything will be good

markjfine, 2023-07-24:
Have a look at [Homebrew](https://brew.sh/). It's a good package manager that can add all the benefits of Linux that macOS/FreeBSD doesn't provide, plus is helps keep everything updated (including Python and Python packages) in one shot. If you're already using a package manager, perhaps it already provides an alternative to macOS's openSSL.

ControlAltAndDelete, 2023-07-24:
openSSL is installed but the search still somehow fails and I've made sure openSSL 3.1 is installed and is in /opt

markjfine, 2023-07-24:
If you open a terminal and type "openSSL version" and it still sees LibreSSL, then you need to make sure something's in the path so it sees the /opt one first.

ControlAltAndDelete, 2023-07-25:
it reports

OpenSSL 3.1.1 30 May 2023 (Library: OpenSSL 3.1.1 30 May 2023)

but the search still errors, is there something wrong with my python install or pip or even homebrew?

markjfine, 2023-07-25:
If it's not OpenSSL, then I'm going back to my original assessment that something in the firewall is blocking it from accessing the internet. We've exhausted any other possibility. Everything is working in Ubuntu and we've eliminated any other virtual difference between the two.

ControlAltAndDelete, 2023-08-13:
[image]

It fixed itself. It must be something with pip3, brew, or anything with the packages. The only thing is the bookmarking bug and the weird retrieval id error in the log.

ControlAltAndDelete, 2023-08-13:
Also, if i can make a recommendation after all those bugs is like on startup or via enabling a setting to automatically clear the traffic and weather maps so inaccurate information isn't displayed and that weather map animations arent displayed weirdly due to information from two different time frames

ControlAltAndDelete, 2023-08-13:
There is a small issue where after a while it would stop searching for album art for some reason (musicbrainzngs)

markjfine, 2025-05-09:
Closing this as apparently completed.

closed 2025-05-09 by markjfine