markjfine/nrsc5-dui #36: GUI fails sporadically

gstalnaker, 2023-11-22:
Linux HOLLIN 6.1.0-10-amd64 https://github.com/theori-io/nrsc5/pull/1 SMP PREEMPT_DYNAMIC Debian 6.1.38-2 (2023-07-27) x86_64 GNU/Linux

MX Linux MX-23.1, KDE Plasma 5.27.5, Debian 12.2

NRSC5-DUI 2.2.2, newly downloaded and installed.

nrsc5 revision https://github.com/theori-io/nrsc5/commit/80ab64eeb2ba3378ed1ef7ef62508176329497ca, RTL-SDR v3 receiver, using the dipole attenna.

Often, but not everytime, the nrsc5-dui gui will crash. The nrsc5 process it spawned to play the SDR feed continues. As I type this, the GUI is gone, yet I'm hearing the Channel. This is the output for this event (note that I'm starting it in the background, which is what the &! is for, a ZSH shell command; the behavior occurs whether I do this 'start in the background' or not):

```
*[master][~/bin/nrsc5-dui]$ ./nrsc5-dui.py &!
*[master][~/bin/nrsc5-dui]$ Using Pillow v9.4.0
['nrsc5', '--dump-aas-files', '/home/guyst/bin/nrsc5-dui/aas', '89.90000000000002', '0']

*[master][~/bin/nrsc5-dui]$ ['nrsc5', '--dump-aas-files', '/home/guyst/bin/nrsc5-dui/aas', '104.1', '1']
**
Gtk:ERROR:../../../gtk/gtkcssinheritvalue.c:33:gtk_css_value_inherit_free: code should not be reached
Bail out! Gtk:ERROR:../../../gtk/gtkcssinheritvalue.c:33:gtk_css_value_inherit_free: code should not be reached
**
Gtk:ERROR:../../../gtk/gtkcssinheritvalue.c:33:gtk_css_value_inherit_free: code should not be reached
Bail out! Gtk:ERROR:../../../gtk/gtkcssinheritvalue.c:33:gtk_css_value_inherit_free: code should not be reached

*[master][~/bin/nrsc5-dui]$ ps | grep nrsc5 
guyst    3676075       1  3 20:38 pts/14   00:00:00 nrsc5 --dump-aas-files /home/guyst/bin/nrsc5-dui/aas 104.1 1
guyst    3676385  864985  0 20:38 pts/14   00:00:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn --exclude-dir=.idea --exclude-dir=.tox nrsc5
```

I can restart nrsc5-dui, but it does not recognize that nrsc5 is currently running and was started by it before it crashed.

argilo, 2023-11-22:
Pure speculation: this sort of mystery error can sometimes happen if GUI operations are performed from outside the GUI thread. At a glance it looks like that could be happening in some places (e.g. `play` -> `parseFeedback` -> `processWeatherOverlay` -> `updated`, `play` -> `parseFeedback` -> `processTrafficMap` -> `updated`, `play` -> `parseFeedback` -> `on_btnStop_clicked`). If it's possible to get a stack trace somehow (running inside `gdb`?), that would help narrow down the problem.

gstalnaker, 2023-11-22:
I'll try that tomorrow.

markjfine, 2023-11-22:
I was also going to suggest switching on the logging feature in Settings if it's not already set. That may also give a clue where it's happening.

What also may be helpful is if I can get a snapshot of the Settings tab, so I know what other features are being used. The musicbrainz feature that attempts to download album art that is not provided by the station can sometimes be a problem. 

I mostly run this on a Mac, but I'll run it on Fedora 39 to see if there's a similar problem there.

markjfine, 2023-11-22:
I'm still checking in Linux, however, as I was preparing it, something else occurred to me: Admittedly, the formal release 2.2.2 is quite old and several changes have been made since then. I've just been too busy to create an updated release package. What you might do is pull down the latest nrsc5-dui.py from the repository and replace the one from v2.2.2, as well as to update some of the Python requirements listed in requirements.txt with pip3 and see if it doesn't improve things. In particular, I see you're using Pillow 9.x. That really should be updated to 10.x. That's where all the image processing is for album art and station logos, as well as traffic and weather images.

gstalnaker, 2023-11-22:
Update - running from gdb (now I know how) produces no issues in this morning's testing (multiple runs). Of course LOL I'll keep working at it to try and get you useful data.

Q: what is the gdb command that will save such useful data for upload to this report?

I'll review your other comments. I'm happy to run a nightly, though I thought that's what I was running.

And, while I have you, thanks to you and all the developers who've worked on this application. I love finding something unexpected that gives me something I've missed. I moved into a condo that is about 200 yards from a local television station and its antenna. My condo is also built into a ridge, with the station SE of me, and my unit faces west. So, I've my whole building and the ridge between me and it. This situation causes lots of radio/em issues. Though I live easily inside the 5G AT&T coverage area, I only get 1 bar of cell reception and no 4G, 4G LTE, or 5G at my unit. That ridge and building between me and the tower where the cell antennas are mounted. But being so close to that tower means that radio reception is awful because of bleed over. Interestingly, the one radio that gets good reception is an old Sony handheld "transistor" radio. All of my fancy radios, including my Pioneer AV Receiver, get awful reception, even with powered antennas. But this SDR? Amazingly, the software filters, etc. just mostly work. I've mapped nearly 2 dozen stations, and with HD nine additional channels, I can listen to. Amazing!! So, thank all of you!!

markjfine, 2023-11-22:
You're certainly welcome and someone's already working to make this a lot better as I've been quite busy with other things over the past year. One thing I can say is that the new interface, which will be Qt-based, looks just fantastic.

I too echo the frustration of living in an area with poor cell reception.

As for gdb - and it's been a long time since I've used it. That said, if I'm not mistaken, I think you can just pipe the output that normally goes to the screen to a file.

gstalnaker, 2023-11-22:
1. I was using the latest, and I did update the packages. BUT ... I was using the wrong python. I have a python v3 venv (I know) because of issues with another application. I needed to set the #! correctly to run in the venv (for me /home/guyst/bin/python). I am now doing that and I see that Pillow v10.0.0 is in use as you indicated. Well, I say I'm using the latest - I used the gh cli command to git the code and install it. I don't see any alternative on the repository. The app still shows 2.2.2 as the version.
2. I can pipe the output (at the least I know how redirect works LOL). Assuming, of course, that it fails again. You know that goes.
3. Settings screenshot attached.
[image]
4. Better Arsenal than ManU.

gstalnaker, 2023-11-22:
Welp. Added a print line to the py code to determine if running in the venv, and no, it is not:

```
print("In venv? ", sys.prefix == sys.base_prefix)
```
Output:
```
"In venv? False"
```
So don't consider that in your deliberations.

markjfine, 2023-11-22:
To keep it current, you just need to go to the main directory (in this case, where nrsc5-dui.py is), and run
`git pull origin master --tags`

Settings look fine, and musicbrainz wasn't being used (Download Album Art option) so that should be ok. I think once you get the Python sorted you should be ok, but let me know if there continues to be any issues.

And yes, better Arsenal than ManU... My daughter has a new boyfriend that has DC United season tickets. I asked her to ask him who he supports and to answer very carefully...

gstalnaker, 2023-11-22:
```
[master][~/src/nrsc5-dui]$ git pull origin master --tags
From https://github.com/markjfine/nrsc5-dui
 * branch            master     -> FETCH_HEAD
Already up to date.
```
Will update here when I have something. Thus far today it hasn't horked once. l

gstalnaker, 2023-11-22:
Update:

nrsc5 and/or nrsc5-dui issue.

What I hear.

Normal output, then, in this specific instance, Frank Sinatra singing "Or she could laugh" that repeats every 12 or so seconds.

What I see.

Then the nrsc5-dui GUI freezes, is unresponsive, then redraw fails (see image):

Forced to pkill it.

Log output extract from within nrsc5-dui:
```
15:07:50 Title: The Lonely Bull
15:07:50 Artist: Herb Alpert & The Tijuana Brass
15:07:50 XHDR: 0 BE4B7536 10846
15:07:50 Audio bit rate: 30.7 kbps
15:07:51 Country: US, FCC facility ID: 41980
15:07:51 Station name: WZEE-FM
15:07:51 Slogan: Z-104 All The Hits
15:07:51 Message:  
15:07:51 Station location: 37.755005, -122.452148, 3776m
15:07:51 Audio program 0: public, type: Top 40, sound experience 0
15:07:51 Audio program 1: public, type: Classical, sound experience 0
15:07:51 Audio program 2: public, type: Religious Music, sound experience 0
15:07:51 Data service: public, type: Navigation, MIME type 469
15:07:51 Data service: public, type: Image Maps, MIME type 2d7
15:07:51 Data service: public, type: Image Maps, MIME type e96
15:07:51 Data service: public, type: Navigation, MIME type eb2
15:07:52 MER: 7.8 dB (lower), 11.1 dB (upper)
15:07:52 BER: 0.004137, avg: 0.002814, min: 0.000000, max: 0.011485
15:07:52 Stream data: port=0880 seq=9AD3 mime=B39EBEB2 size=847
15:07:52 Stream data: port=0880 seq=9AD4 mime=B39EBEB2 size=263
15:07:52 Stream data: port=0881 seq=9E8A mime=4EB03469 size=10
15:07:52 Title: The Lonely Bull
15:07:52 Artist: Herb Alpert & The Tijuana Brass
15:07:52 XHDR: 0 BE4B7536 10846
15:07:52 Audio bit rate: 31.7 kbps
15:07:52 Country: US, FCC facility ID: 41980
15:07:52 Station name: WZEE-FM
15:07:52 Slogan: Z-104 All The Hits
15:07:52 Message:  
15:07:52 Station location: 37.755005, -122.452148, 3584m
15:07:52 Audio program 0: public, type: Top 40, sound experience 0
15:07:52 Audio program 1: public, type: Classical, sound experience 0
15:07:52 Audio program 2: public, type: Religious Music, sound experience 0
15:07:52 Data service: public, type: Navigation, MIME type 469
15:07:52 Data service: public, type: Image Maps, MIME type 2d7
15:07:52 Data service: public, type: Image Maps, MIME type e96
15:07:52 Data service: public, type: Navigation, MIME type eb2
15:07:53 MER: 9.5 dB (lower), 12.5 dB (upper)
15:07:53 BER: 0.002118, avg: 0.002808, min: 0.000000, max: 0.011485
15:07:53 Stream data: port=0880 seq=9AD5 mime=B39EBEB2 size=1008
15:07:53 Stream data: port=0880 seq=9AD6 mime=B39EBEB2 size=839
15:07:53 LOT file: port=0802 lot=10846 name=WZEE022a5e.jpg size=6248 mime=1E653E9C expiry=2023-11-22T21:36:00Z
15:07:53 Title: The Lonely Bull
15:07:53 Artist: Herb Alpert & The Tijuana Brass
15:07:53 XHDR: 0 BE4B7536 10846
15:07:53 Audio bit rate: 29.9 kbps
```
To tshoot, run nrsc5 from terminal. Console output extract:
```
15:09:49 Title: The Lady Is A Tramp
15:09:49 Artist: Frank Sinatra
15:09:49 XHDR: 0 BE4B7536 10365
15:09:49 Title: The Lady Is A Tramp
15:09:49 Artist: Frank Sinatra
15:09:49 XHDR: 0 BE4B7536 10365
15:09:49 Audio bit rate: 30.6 kbps
15:09:50 Country: US, FCC facility ID: 41980
15:09:50 Station name: WZEE-FM
15:09:50 Slogan: Z-104 All The Hits
15:09:50 Message:  
15:09:50 Station location: 37.755005, -122.452148, 576m
15:09:50 Audio program 0: public, type: Top 40, sound experience 0
15:09:50 Audio program 1: public, type: Classical, sound experience 0
15:09:50 Audio program 2: public, type: Religious Music, sound experience 0
15:09:50 Data service: public, type: Navigation, MIME type eb2
15:09:50 Data service: public, type: Navigation, MIME type 469
15:09:50 Data service: public, type: Image Maps, MIME type 2d7
15:09:50 Data service: public, type: Image Maps, MIME type e96
15:09:51 MER: 9.4 dB (lower), 13.9 dB (upper)
15:09:51 BER: 0.001426, avg: 0.004761, min: 0.001426, max: 0.012744
15:09:51 Stream data: port=0880 seq=9B61 mime=B39EBEB2 size=1110
15:09:51 Stream data: port=0881 seq=9EFF mime=4EB03469 size=31
15:09:51 Stream data: port=0881 seq=9F00 mime=4EB03469 size=165
15:09:51 LOT file: port=0810 lot=990 name=TMT_0358l0_2_3_20231122_2104_03de.png size=6783 mime=4F328CA0 expiry=2023-11-22T21:36:00Z
15:09:51 Title: The Lady Is A Tramp
15:09:51 Artist: Frank Sinatra
15:09:51 XHDR: 0 BE4B7536 10365
15:09:51 Audio bit rate: 30.7 kbps
15:09:51 Country: US, FCC facility ID: 41980
15:09:51 Station name: WZEE-FM
15:09:51 Slogan: Z-104 All The Hits
15:09:51 Message:  
15:09:51 Station location: 37.755005, -122.452148, 3488m
15:09:51 Audio program 0: public, type: Top 40, sound experience 0
15:09:51 Audio program 1: public, type: Classical, sound experience 0
15:09:51 Audio program 2: public, type: Religious Music, sound experience 0
15:09:51 Data service: public, type: Navigation, MIME type eb2
15:09:51 Data service: public, type: Navigation, MIME type 469
15:09:51 Data service: public, type: Image Maps, MIME type 2d7
15:09:51 Data service: public, type: Image Maps, MIME type e96
```
Then same 12-second repeat. But note the difference in the outputs. The second correctly shows Artist: Frank Sinatra. But the first shows Herb Alpert & The Tijuana Brass, which is interesting. If I let it run, the INFO in the stream changes, but the audio is stuck on this repeating fragment, about 1-2 seconds.

This fragment is persistent across runs of both nrsc5 standalone and nrsc5-dui. Could be something at the source, but figured I'd check here as well.

Is there a cache somewhere that I've horked with my starts and pkill stops? Or something I need to delete when I do that?

markjfine, 2023-11-22:
Interesting.

Usually what nrsc5 reports is correct, so the Tijuana Brass thing could just be an info delay at the station. I've seen that in a few instances.

The audio is a different issue. Just wondering if there's something in your audio interface that's not quite right (I've not trusted anything ALSA-based for decades). Thinking this may be an additional/independent processing problem to what you were reporting here as well as over at nrsc5.

Just curious: Did you build nrsc5 using either the NEON or SSE options, and, what architecture are you running?

gstalnaker, 2023-11-22:
Architecture in the first post. I used no additional build options, so built with the defaults.

And just to confirm, I just ran

```
*[master][~/bin/nrsc5-dui]$ nrsc5 104.1 1 
```

The audio for this frequency and channel is still glitched. I can run 
```
nrsc5 104.1 0
```
and shift between the three channels for this station, and it's only channel 1 that exhibits this glitch despite the updated INFO.

If there is no cache or anything that can be local, I'd assume this must be at the source, right?

I didn't mean, by putting this in this report that it's related to our previous posts. I perhaps should not have mentioned it. But thought it easy to ask if there were local caches, etc. I could easily delete.

markjfine, 2023-11-22:
`I didn't mean, by putting this in this report..`

No problem. nrsc5-dui does no caching, but nrsc5 might have some buffering in order to make the packet streaming seamless.

markjfine, 2023-11-23:
Incidentally, Most of the year's gone by and I've been remiss in updating the versioning and dates in the py file. I've just fixed that.

gstalnaker, 2023-11-23:
FYI. While exploring use of pulseaudio to stream the audio from ncrs5 and SDR++, I horked pulseaudio. To fix that I attempted to upgrade/reinstall packages using MXLinux's package tool. The system now will not boot. I may revert to kubuntu. Will be a few days to get this fixed.

markjfine, 2023-11-24:
Ugh. Good luck.

closed 2025-05-09 by markjfine