markjfine/nrsc5-dui #39: Images/Maps not displaying at all on current version

dragonbytes, 2024-05-13:
Hi Mark, so I was trying to get the current version working on my system and for some reason, no album art, station art, or maps are ever displayed. I went back to an old version 2.2.0 and it works fine there, but not on the current. I am using Python 3.10 and tried it both with Pillow 9.0.x and 10.x. The app otherwise works perfectly, just no image displays. What am I doing wrong? lol.

Also, side-question, i'm trying to get my feet wet with visual studio coding and figured as a learning experience, id try my hand at a nrsc5-dui-inspired Windows .NET frontend.  How the heck do you get the weather maps to work? As far as I can tell, nrsc5 LOT files only provide the transparent Doppler layer. Are you pulling the actual background map from the network? Thanks as always for all your work on this project!

markjfine, 2024-05-13:
I'd need to know what settings you're using and on what machine/OS. Since you've mentioned .NET, I'm guessing you're on Windows? If that's the case, there may be an issue with using any spaces in the path to the `aaa` directory, though I'm not sure why that would change between 2.2 and anything later.

dragonbytes, 2024-05-13:
Ahhh!! I didn't consider you might have a local map of the whole country. I figured that would be way too big in size to be practical, but it compresses really well it seems!

As far as my setup, the Windows comment was unrelated to my issue. Here is the system/setup I used to try your current version:

System: PC Intel 13700K x64 platform
OS: Kubuntu 22.04 (KDE desktop)
Python Version: 3.11
Pillow: 10.x (installed via PIP)

The other dependencies were installed using PIP. PyGObject was already preinstalled on the distro. If theres any other
details you need, please let me know.

markjfine, 2024-05-13:
Just looking at the code, the only things that changed regarding album art, etc. was to handle a deprecation issue in pillow regarding scaling. Very odd that it's not displaying things.

dragonbytes, 2024-05-13:
Is there a specific version of Pillow that is required? Is there perhaps a debug mode I could enable to get a more verbose description of whats going on during execution?

markjfine, 2024-05-13:
Pillow 10.3.0 is currently what I use.
You can check Log to File in Settings.

dragonbytes, 2024-05-16:
Do you mind if I ask how you were able to convert the Weather Radar coordinates to the correct pixel offsets within Map.png without a point-of-reference? I don't see any definition of which coordinates the upper-left part of the map of the USA correspond to.

markjfine, 2024-05-16:
If I remember correctly, the weather information includes a lookup file (DWRI). The filename is encoded with a UTM-like Doppler Weather Radio Area ID so the overlay data can be referenced. For example, the code for the Washington DC area is `03ggta`, and the lookup file would be titled something like `xxxx_DWRI_03ggta_revxx_xxxx.txt` in the `aas` directory. Inside that file are the coordinates to create anther file `BaseMap_03ggta.png` in the `map` directory. For example, the line you're looking for would be:
```
Coordinates="(40.90135,-79.98062)";"(36.70309,-74.59073)"
```
The Base Map is then carved from `map.png` using hard-coded calculations and general pillow image routines, as taken from the original nrsc5-gui code. I've done very little to modify that part other than minor pillow maintenance.

dragonbytes, 2024-05-28:
I'm still working on troubleshooting this issue. I tried enabling the "Log to file" option, however all that captures is the output for NRSC5 and doesn't include any of YOUR debug info I see in the script file. Is there a command-line argument I can use to output your internal debug messages or would I need to learn how to use a python debugger to see them?

Also, thanks alot for your help with the weather map. It was a HUGE help!! I notice that none of the GUI projects ever mention "HD Radio" by name or include any of the official logos. Could we get in trouble for use those even though these are completely open-source not-for-profit projects?

markjfine, 2024-05-29:
Those debug lines (at least the ones I added) should just show in the terminal window you ran it on.

I don't think there's an issue using any such logo as long as no fees are being requested. Otherwise their lawyers can claim the app is making money off their images. Pretty sure that's true for any logo.

dragonbytes, 2024-06-06:
So I finally found a configuration that properly shows the various artwork/maps etc with the latest version. Up until now, I was using PIP to install all the dependencies, this time I used APT to install them instead (and only installed the dependencies that were complained about when trying to run). I'll list everything with version numbers below:

Kubuntu 24.04 LTS
Python 3.12.3
Pillow 10.2.0
Numpy 1.26.4
Dateutil 2.8.2-3
OpenSSL 23.2.0-1
musicbrainzngs 0.7.1-6

Hopefully this provides some useful info/insight. Thanks again for your help!

dragonbytes, 2024-06-07:
I guess I can just mark this as closed? Also, would you mind if I reached out to you directly in the future? I'm a fellow shortwave/ham radio listener and programmer. Maybe we could collab on something! :)

markjfine, 2024-06-07:
I can close it. And yes, we can collaborate on things if you like. I've not been doing much developing of anything nor radio listening lately. Mostly catching up on some gaming that I've missed.

closed 2024-06-07 by markjfine

dragonbytes, 2024-06-07:
I totally understand that. I tend to go through phases. Was doing alot of retro computer coding and then my brother got me into Fortnite and I became obsessed lol. Now i've been diving back into SDR tinkering and learning modern desktop app development. What's the best way to reach out? Do you have a Discord username?

markjfine, 2024-06-07:
I've updated my profile [here](https://github.com/markjfine), which has my web site, email, Reddit, and Twitter accounts. Though I'm rarely on Twitter anymore. I'm also Gooner7mjf on Discord and Steam.