markjfine/nrsc5-dui #42: Cannot bookmark a subchannel without switching to the subchannel, stopping, and starting.

jg303, 2024-07-11:
Environment: RPi3 with Raspbian, RTL-SDR Blog v3 device.

Issue: Cannot bookmark a subchannel without switching to the subchannel, stopping, and starting.

Steps to replicate: Tune to 93.3 HD-1 (as an example, this is for any frequency). Bookmark the subchannel. Cannot bookmark anything else. If I stop and start on 93.3 HD-1, I still cannot bookmark anything else. Must start on a non-bookmarked frequency & subchannel to bookmark another subchannel. Once that frequency & subchannel are bookmarked, I must select another unbookmarked frequency & subchannel to continue.

Current workaround: Just following the steps above, or personally modifying the JSON.

Thank you for this software, it has otherwise performed flawlessly!

markjfine, 2024-07-11:
I suspected there might be a weird timing/precedence issue with bookmarking that I never corrected in the original nrsc5-gui.

Unable to get to it right now but I'll have a look with a fix in a few days.

markjfine, 2025-05-09:
This should be fixed using the latest change to nrsc5-dui.py in master.

closed 2025-05-09 by markjfine