markjfine/nrsc5-dui #44: Unable to install dependencies on Ubuntu (PEP 668 error)

robman501a, 2024-12-02:
How the heck do you install the dependencies? I do pip install -r <path_to requirements.txt> as stated but I keep getting the PEP 668 error.
[image]
I've installed the Python3-full package

robman501a, 2024-12-02:
I just want to add that it's a fresh installation on Ubuntu 24.04.1 LTS, so I might be missing some packages.

markjfine, 2024-12-03:
Not gonna lie. I have no idea what you did or need to do to fix it.

That said, I've had problems using pip (or pip3) in the past as python versions change. pip tends to want to install packages into only the currently used version. So when the version changes to say, 12 to 13 anything new goes into 13, while 12 is left holding the bag. Looks like you may be running 12.

What I would recommend is to just manually install each package listed in requirements.txt using something like dnf or apt (which the error recommends), which tends to install across all installed versions of python. The downside is that you may have to search for the correct package nomenclature, as they change from distribution to distribution: For example, my Fedora packages may be named differently that your Ubuntu ones.

robman501a, 2024-12-03:
You said that you use Fedora? Please list all the packages that you installed before you compiled this program. Maybe I'm missing something. I'm trying a different Ubuntu but I'm seeing new errors, like meson isn't found, libcario.... etc.

I've tried pip and pip3 and managed to get this far....

rob@rob-Aspire-A315-41:~/nrsc5-dui$ sudo pip3 install -r ~/nrsc5-dui/requirements.txt
Collecting pygobject==3.50.0
  Using cached pygobject-3.50.0.tar.gz (1.1 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [28 lines of output]
      + meson setup /tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6 /tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6/.mesonpy-d3ymdjmm -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Dtests=false -Dwheel=true --wrap-mode=nofallback --native-file=/tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6/.mesonpy-d3ymdjmm/meson-python-native-file.ini
      The Meson build system
      Version: 1.6.0
      Source dir: /tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6
      Build dir: /tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6/.mesonpy-d3ymdjmm
      Build type: native build
      Project name: pygobject
      Project version: 3.50.0
      C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
      C linker for the host machine: cc ld.bfd 2.38
      Host machine cpu family: x86_64
      Host machine cpu: x86_64
      Program python3 found: YES (/usr/bin/python3)
      Found pkg-config: YES (/usr/bin/pkg-config) 0.29.2
      Run-time dependency python found: YES 3.10
      Run-time dependency gobject-introspection-1.0 found: YES 1.72.0
      Run-time dependency glib-2.0 found: YES 2.72.4
      Run-time dependency gobject-2.0 found: YES 2.72.4
      Run-time dependency gio-2.0 found: YES 2.72.4
      Run-time dependency gmodule-2.0 found: YES 2.72.4
      Run-time dependency libffi found: YES 3.4.2
      Run-time dependency cairo found: YES 1.16.0
      Run-time dependency cairo-gobject found: YES 1.16.0
      Message: Found pycairo with target Python: /usr/lib/python3/dist-packages/cairo/include
      
      ../meson.build:82:22: ERROR: Include dir /usr/lib/python3/dist-packages/cairo/include does not exist.
      
      A full log can be found at /tmp/pip-install-tan4rm_1/pygobject_09b399f945394954a221ba494f6ba4e6/.mesonpy-d3ymdjmm/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

markjfine, 2024-12-03:
First Python isn't compiled, it's interpretive.

Second, everything it needs is listed in requirements.txt.

Third, I've already told you that pip is unreliable. Please use apt install.

Other than this, I don't know how to help you.

robman501a, 2024-12-03:
Okay, thanks. I'm going to move to Fedora and see if I have any luck there.
I've manually installed the packages with apt install but now I'm getting this.

rob@rob-Aspire-A315-41:~/nrsc5-dui$ python3 nrsc5-dui.py
Using Pillow v9.0.1
Traceback (most recent call last):
  File "/home/rob/nrsc5-dui/nrsc5-dui.py", line 34, in <module>
    imgLANCZOS = Image.Resampling.LANCZOS
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 65, in __getattr__
    raise AttributeError(f"module '{__name__}' has no attribute '{name}'")
AttributeError: module 'PIL.Image' has no attribute 'Resampling'

markjfine, 2024-12-04:
PIL is pillow. You didn't install it, even though it was on the list.

Why do you think just swapping around linux distros will solve your problem instead of simply reading and following instructions?

robman501a, 2024-12-08:
Hi, so I've successfully compiled NRSC5-DUI on Fedora. I haven't tested Ubuntu... yet. Will try that later in a virtual machine.

markjfine, 2024-12-09:
Again, NRSC5-DUI doesn't compile. It's a Python script.

Have just realised that there are several problems with running under Python v3.13. Some of the dependencies aren't ready for it yet, namely dateutil. Until dateutil (and others) are properly built for 3.13, recommend running no later version than Python 3.12.

If it's working under Fedora, there's no need to test under Ubuntu (under a VM or otherwise) as I'm certain there are mismatched dependencies there. I'd also prefer you did not test anything under a VM, because nrsc5 itself requires real-time USB interfacing that can (and usually will) be interrupted when running under a VM. Bottom line: it won't work correctly.

markjfine, 2025-05-09:
Is this still valid or should the issue be closed?

robman501a, 2025-05-09:
Go ahead and close this issue. I haven't touched this in some time, also I haven't really played around with Linux as I mainly use Windows.

closed 2025-05-09 by robman501a