markjfine/nrsc5-dui #45: Problem with running NRSC5-DUI on Debian 12 XFCE

Route66Fan, 2025-04-02:
When I try running NRSC5-DUI on Debian 12 XFCE, the app will launch & will try sending a command to NRSC5 when I tune to a known frequency that has HDRadio sidebands & hit play, however there will be no audio & the bar at the bottom of the NRSC5-DUI app will show 0.0kbps, 0.0db & 0.00% BER. 

I compiled the version of NRSC5 for the RTL-SDR with the cmake option -DLIBRARY_DEBUG_LEVEL=1. 

NRSC5 is installed in my /usr/local/bin/ directory. 

Also, for some reason, I have to run NRSC5-DUI as root or the app fails to run entirely. 

Please help, thank you!

markjfine, 2025-04-02:
Unfortunately I'm not familiar with Debian, only Fedora. I'm going to guess a lot of this may have to do with with how Debian installs Python, what version is of Python is being used, and the dependencies. I'm currently running Python 3.13 aliased as Python3 under Fedora and used dnf instead of pip3 to maintain the dependencies.

Secondly, I would look to see if you can run nrsc5 via command line in a terminal to see if it can be run without being root and if it can play the station you describe.

Also, I use a script to kick off nrsc5-dui under Fedora as follows:

```
#!\/bin/sh
export PATH=/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/x11/bin
cd <path to where you have nrsc5-dui installed>:python3 nrsc5-dui.py
```

You may want to see if something like this works better for you once you modify the cd line with your path to nrsc5-dui.py. This ensures that everything is reachable in the path, including all of the Python parts and is running from the correct starting location.

Route66Fan, 2025-04-03:
I'll try running it in Fedora, thank you!

Route66Fan, 2025-04-03:
I forgot to mention that I was running Debian in a VM. 

I've tried installing it in another VM running Fedora with the RTL-SDR connected to that & I ended up having the same problem there as well. 

I think that the issue may be caused by the RTL-SDR driver not being installed. 

I tried installing the RTL-SDR drivers using the command "sudo dnf install rtl-sdr- gr-osmosdr-" & got the following error: 
Failed to resolve the transaction:
No match for argument: rtl-sdr-
No match for argument: gr-osmosdr-
You can try to add to command line:
  --skip-unavailable to skip unavailable packages

markjfine, 2025-04-03:
> I forgot to mention that I was running Debian in a VM. 

Well that would certainly explain a lot. Computing-wise, a VM works well provided it recognizes the usual hardware involved (e.g., video, keyboard, etc.). You are introducing a USB device that requires specific USB drivers for Windows, let alone Linux.

Take a look in the Readme about the problems I had getting a rtlsdr to run correctly under other posix-compatible, minimal, and transparent VMs, such as mingw and MSYS2. It's not trivial because you're still fighting a Windows-controlled hardware layer to even get to the device via USB.

Let's suppose you even make contact with the device and get things to initially work. Then you have to also deal with an unsteady stream of data because of the VM interrupting the USB and killing things.

All of this is why I asked you to take the steps I outlined first, because it would isolate if you could even run nrsc5 from a command line, which I don't think may be possible given your environment. That is all nrsc5-dui does, is spawn off nrsc5 with the appropriate command line switches, and occasionally allow to pipe HD sub-channel changes and/or quit nrsc5. If you can't run nrsc5 from a terminal window in Linux, then nrsc5-dui isn't going to do anything other than sit there.

What I would advise you to do if you're interested in doing Linux things on an Intel box, is to not mess with VMs at all. Make the box dual-booting as Linux/Windows. Then you can install everything under Linux (Debian, Fedora, Ubuntu, whatever) and not worry about Windows' non-posix layers and general incompatibilities screwing everything up.

Route66Fan, 2025-04-03:
Ok, I'll think about that. 

Thank you for your help!

closed 2025-04-03 by markjfine