markjfine/nrsc5-dui #47: Issues with Traffic Maps and Weather Maps

ControlAltAndDelete, 2025-06-15:
Downloading the new version of nrsc5-dui on my M3 Air was a breeze. Upon discovering the HERE Images support, I immediately downloaded the new files and a new version of NRSC5.

Here's the sitch:
1. When tuning into a HERE station for the 1st time (not tuning into any other TTWN or HERE station), traffic maps will download normally and without question but,
2. If you tune to a TTWN station afterwards, a traffic map will download successfully but tuning to another HERE station will result in new traffic maps from HERE not downloading, choosing to skip images. 1 --> 5. Haven't checked if going back to another iHeart station will replace the image or if it's just WMMR not working properly.

TTWN: [image]
HERE: [image]
HERE map successfully generated before tuning into a TTWN station:  [image]

4. Weather Maps do not work with HERE. It could be a problem with HERE itself. TTWN stations do correctly get Weather Maps but HERE does not. Just a green image. Also, tuning into a HERE station for Weather Maps (without previously tuning into a TTWN one), results in no base map being generated. You have to first tune into a TTWN station for a base map and even then creates a weird grey background. 
TTWN: [image]
Image received from the HERE station from NRSC5-DUI: [image]

5. After getting a base map generated from tuning to a TTWN station, the HERE base map generated is screwed up. Just a green background, and zoomed into a random square.
[image]

I can discuss this further and record videos if it fits. That's just the behavior I am seeing.

ControlAltAndDelete, 2025-06-15:
[image]

Creating a fresh instance without files, it seems like at another station the green images occur. It might be an NRSC5 fault or station fault but this at another HERE station. The station is farther away and weak though.

Here's what happens with Weather Maps
[image]

markjfine, 2025-06-16:
The HERE traffic maps themselves are better looking that the TTWN, but didn't realize they didn't contain much data. Looking at the HERE map, I'd have no issue driving into center city.

ControlAltAndDelete, 2025-06-16:
[image]

Update: base maps did create without the need to tune to a TTWN station first but they're still using a small 200x200 square of OSM imagery.

Base Map: [image]
Weather Overlay with Base Map: [image]

markjfine, 2025-06-16:
There's no need to tune to a TTWN station first. As I mentioned before, the coordinates provided by HERE are trash so I need a full set of traffic tiles with the same timestamp and ID to get the right ones in order to generate a basemap for that ID. You can feasibly get a weather overlay after traffic tile 9 is received, but it will ignore it if it hasn't seen a corresponding tile 1. What probably happened is that you tuned away from the HERE station then came back, got tile 1, and the appropriate basemap was generated.

ControlAltAndDelete, 2025-06-17:
I also noticed that if you tune in to another station in another market with HERE, nrsc5-dui will only show the original market weather maps generated. I tuned to Philly stations and got 6 weather maps, tuning to a Trenton station and getting 1 weather map. The program seems to ignore the new weather maps and will refuse to show it if it's from a new coordinate, opting to show previous weather maps from an old coordinate. This behavior does not seem to impact Traffic Maps.

Also of note, you may want to test this in your market but Doppler Radar .pngs are no longer fully green, they're now properly transparent and can be overlaid on base maps in a future update. (Tested on WBEB in Philly and WKXW in Trenton).

markjfine, 2025-06-17:
Like before, even if it's a new market, it will still show whatever weather map was created last, until it can properly display a new one. It takes roughly five minutes to get through a full traffic/weather cycle. If you're not dwelling on that station long enough to get the data, it's not going to show you anything new. That was even true for TTWN.

markjfine, 2025-07-30:
Closed this issue because there's really nothing here to fix. Most of the problems are caused by HERE itself or the station's implementation of HERE data.

closed 2025-07-30 by markjfine